peng-zhihui/HelloWord-Keyboard #8: 请问键盘支持热插拔吗？谢谢

Dr-Huang2006, 2022-08-01:


GongXiaoBei77, 2022-08-01:
这是当然的,请认真去看介绍视频

git-flyer, 2022-08-01:
啊，这，USB设备理所当然支持吧

Dr-Huang2006, 2022-08-01:
我的意思是键盘的轴支不支持热插拔，不是usb支不支持

GongXiaoBei77, 2022-08-01:
> 我的英文是键盘轴支不支持热封，不是的usb支不支持

事实上,原视频6分48秒中左右有展示过,而且不止一次出现,甚至在DOCS中也有明确的热拔插轴座的文档.

AkariiinMKII, 2022-08-03:
> 我的意思是键盘的轴支不支持热插拔，不是usb支不支持

请先浏览一遍[项目文件](https://github.com/peng-zhihui/HelloWord-Keyboard/blob/main/5.Docs/1.Datasheet/%E5%87%AF%E5%8D%8E%E7%83%AD%E6%8B%94%E6%8F%92%E9%94%AE%E8%BD%B4%E5%BA%A7.jpg)再提问，谢谢

Mo3et, 2022-08-08:
很明显是可以的 现在基本配合新手入门 都有热插拔轴座
README也有提到 是从Xikii的S98 改的 在这里问 还不如去xikii的群问

@Dr-Huang2006  请尽快自行Close 这个issue 有啥其他讨论去repo的Discussions里 这里主要是项目开发内容的issues 不要滥用issues！
（下面是Readme的引用）
> 注意：Issues里面是讨论项目开发相关话题的，不要在里面发无意义的消息，不然watch了仓库的人都会收到通知邮件会给别人造成困扰的！！！灌水可以在仓库的Discuss里讨论！

closed 2022-08-08 by Dr-Huang2006