peng-zhihui/HelloWord-Keyboard #71: 新问题，为什么吧任何键赋零层的0值就失效了

xiaohaha23, 2022-08-15:
现在问题源就是，零层的0值，无论赋予任何一个1层映射都无效，但当切换为二层映射就有效，初步怀疑是一层0值无意义导致，得通过二层去触发一层的映射，硬件检查无任何问题

peng-zhihui, 2022-08-19:
映射是这样的，0层是必然会执行的，后续的1~N层只会执行一个，也就是说你选的2层那么最终就是先映射0层，然后映射2层，所以你描述的这个问题应该是代码有点小bug，晚点有空我调一下

0xCC0x01, 2022-08-23:
pageUP的bug，应该跟按键定义有关：

1. keymap[KEYMAP_NUM][IO_NUMBER]，有5层映射，每层88个（82个按键+6个Touchpad）
2. 但在定义每层映射时，实际只有83个值，后面的几个默认初始化了0
3. 默认给的0值，恰好是pageUP对应的IO pin

可以尝试下，把keymap 0层映射，最后面的几个值补充完整，应该能搞定，比如：
int16_t keyMap[KEYMAP_NUM][IO_NUMBER] = {
// The first layer, used for aligning 74HC165 IO pins to PCB key layout
{67, 61, 60, 58, 59, 52, 55, 51, 50, 49, 48, 47, 46, 3,
80, 81, 64, 57, 62, 63, 53, 54, 45, 44, 40, 31, 26, 18, 2,
19, 70, 71, 66, 65, 56, 36, 37, 38, 39, 43, 42, 41, 28, 1,
15, 74, 73, 72, 68, 69, 29, 30, 35, 34, 33, 32, 24, 0,
14, 76, 77, 78, 79, 16, 20, 21, 22, 23, 27, 25, 17, 4,
13, 12, 8, 75, 9, 10, 7, 11, 6, 5,
82, 83, 84, 85, 86, 87} //最后的几个值，不要空着

我没硬件，没有尝试，有板子的同学可以试下