peng-zhihui/HelloWord-Keyboard #79: 稚晖君你好，我在复刻瀚文，发现page up键无法使用，检测了下硬件没有问题，可能是程序哪里的bug,感谢回复！

DOCTOROLIVE, 2022-08-18:
page up 键无法使用

peng-zhihui, 2022-08-18:
是一个bug，我晚点调试一下

0xCC0x01, 2022-08-23:
pageUP的bug，应该跟按键定义有关：
1. keymap[KEYMAP_NUM][IO_NUMBER]，有5层映射，每层88个（82个按键+6个Touchpad）
2. 但在定义每层映射时，实际只有83个值，后面的几个默认初始化了0
3. 默认给的0值，恰好是pageUP对应的IO pin

可以尝试下，把keymap 0层映射，最后面的几个值补充完整，应该能搞定，比如：
int16_t keyMap[KEYMAP_NUM][IO_NUMBER] = {
            // The first layer, used for aligning 74HC165 IO pins to PCB key layout
        {67,  61, 60, 58, 59, 52, 55, 51, 50, 49, 48,  47,  46,  3,
            80,           81,    64,    57,    62,    63,    53,    54,    45,    44,    40,    31,    26,    18,        2,
            19,  70, 71, 66, 65, 56, 36, 37, 38, 39, 43, 42,           41,            28,        1,
            15,       74, 73, 72, 68, 69, 29, 30, 35, 34, 33,         32,    24,    0,
            14,         76, 77, 78, 79, 16, 20, 21, 22,    23,     27,    25,          17,       4,
            13,        12,       8,        75,    9,         10,        7,  11,         6,          5,
            82, 83, 84, 85, 86, 87} //最后的几个值，不要空着

JiumiYao, 2022-08-23:
问题具有完全可见的复现能力，看到了楼上[user:[0xCC0x01](https://github.com/0xCC0x01)]的方法补充了映射最后几个值，问题得到了解决。不过在游戏中按键的绑定似乎仍存在问题，测试Steam平台CS:GO和EPIC平台堡垒之夜均无法正常使用点击绑定，不过网页和输入法的确没问题。游戏本身问题待查。

Gralerfics, 2022-08-24:
> pageUP的bug，应该跟按键定义有关：
>
> 1. keymap[KEYMAP_NUM][IO_NUMBER]，有5层映射，每层88个（82个按键+6个Touchpad）
> 2. 但在定义每层映射时，实际只有83个值，后面的几个默认初始化了0
> 3. 默认给的0值，恰好是pageUP对应的IO pin
>
> 可以尝试下，把keymap 0层映射，最后面的几个值补充完整，应该能搞定，比如： [...]

您好，我觉得您的解答非常有道理，应该是与初始化的 0 有关，但是还是不太想得明白为什么会导致 Page Up 失灵。缺少的五个值对应触摸条的后五个触点，如果被初始化为 0 是否应该只是代表多出了五个 Page Up 键（因为 remapBuffer[i] |= 0x80 >> j; 用的是或等于，似乎不会干扰相同功能键的使用）？手上还没有硬件可以尝试，只能空想，希望能得到解答，谢谢 orz。

0xCC0x01, 2022-08-25:
> 您好，我觉得您的解答非常有道理，应该是与初始化的 0 有关，但是还是不太想得明白为什么会导致 Page Up 失灵。缺少的五个值对应触摸条的后五个触点，如果被初始化为 0 是否应该只是代表多出了五个 Page Up 键（因为 remapBuffer[i] |= 0x80 >> j; 用的是或等于，似乎不会干扰相同功能键的使用）？手上还没有硬件可以尝试，只能空想，希望能得到解答，谢谢 orz。

我也没有硬件，只是对着代码看的猜测，发出来也是想请有板子的这位兄弟验证下。

0xCC0x01, 2022-08-25:
你提到的`remapBuffer[i] |= 0x80 >> j`确实是或，不干扰其他的，但后面还有一段hidBuffer相关的代码：
```
index = (int16_t) (keyMap[_layer][i * 8 + j] / 8 + 1); 
bitIndex = (int16_t) (keyMap[_layer][i * 8 + j] % 8);
……

if (remapBuffer[i] & (0x80 >> j))
      hidBuffer[index + 1] |= 1 << (bitIndex);
```
layer1最后的几个值也是0，就相当于在hidBuffer中把keycode为0的按键也放进去了，keycode 0是Reserved。
也就是说按下pageUP时，实际hidBuffer中包含了Reserved以及pageUP，不知道操作系统会如何处理，
如果操作系统认为包含了Reserved的按键信息是无效的，丢弃掉不处理，那pageUP也就无效了

没有验证，纯看代码猜想。

JiumiYao, 2022-08-25:
> 你提到的`remapBuffer[i] |= 0x80 >> j`确实是或，不干扰其他的，但后面还有一段hidBuffer相关的代码：
> 
> ```
> index = (int16_t) (keyMap[_layer][i * 8 + j] / 8 + 1); 
> bitIndex = (int16_t) (keyMap[_layer][i * 8 + j] % 8);
> ……
> 
> if (remapBuffer[i] & (0x80 >> j))
>       hidBuffer[index + 1] |= 1 << (bitIndex);
> ```
> 
> layer1最后的几个值也是0，就相当于在hidBuffer中把keycode为0的按键也放进去了，keycode 0是Reserved。 也就是说按下pageUP时，实际hidBuffer中包含了Reserved以及pageUP，不知道操作系统会如何处理， 如果操作系统认为包含了Reserved的按键信息是无效的，丢弃掉不处理，那pageUP也就无效了
> 
> 没有验证，纯看代码猜想。 还是要等稚晖君debug解决。

您好！您这个方法可以解决了pageup这个问题！（其实几天前看到您评论这个就弄了解决了）但是最后我删评了主要是因为部分游戏还是无法识别pageup，目前我试过了有csgo、apex、堡垒之夜都没办法在设置页面通过点击绑定pageup按键。但是在office、chrome网站和其他我常用的几款软件都是可以的，无法识别的均是游戏，所以当时就先删评想解决这个问题。但是研究了几天还是没啥思路，不知您是否有看法和解决的方案？

JiumiYao, 2022-08-25:
> 你提到的`remapBuffer[i] |= 0x80 >> j`确实是或，不干扰其他的，但后面还有一段hidBuffer相关的代码：
>
> ```
> index = (int16_t) (keyMap[_layer][i * 8 + j] / 8 + 1);
> bitIndex = (int16_t) (keyMap[_layer][i * 8 + j] % 8);
> ……
>
> if [...]

感觉像是驱动的问题？但是这个方面我不太懂，不知从哪里入手。

0xCC0x01, 2022-08-25:
> 您好！您这个方法可以解决了pageup这个问题！（其实几天前看到您评论这个就弄了解决了）但是最后我删评了主要是因为部分游戏还是无法识别pageup，目前我试过了有csgo、apex、堡垒之夜都没办法在设置页面通过点击绑定pageup按键。但是在office、chrome网站和其他我常用的几款软件都是可以的，无法识别的均是游戏，所以当时就先删评想解决这个问题。但是研究了几天还是没啥思路，不知您是否有看法和解决的方案？感觉像是驱动问题，但是不太懂这方面开发，不知从哪里入手。

如你所说，在office、网页等应用pageUP已经可以使用，那我觉得就跟固件没什么关系了。至于游戏无法绑定pageup，这个我不了解，你可以试下其他的键盘是否有同样游戏的问题，个人认为不是键盘固件的问题。

ps: 羡慕你们都已经拿到键盘了，太快了…… 我才上了PCBA的车，东西估计还要几周才能到。

JiumiYao, 2022-08-25:
> 如你所说，在office、网页等应用pageUP已经可以使用，那我觉得就跟固件没什么关系了。至于游戏无法绑定pageup，这个我不了解，你可以试下其他的键盘是否有同样游戏的问题，个人认为不是键盘固件的问题。
>
> ps: 羡慕你们都已经拿到键盘了，太快了…… 我才上了PCBA的车，东西估计还要几周才能到。

哈哈，我朋友是出国在瑞士那边留学，留下来给罗技做工程师了，稚晖君大佬开源了大部分东西，小部分问题他都帮我设计好了。我表弟刚好是干外设厂的，所以制造也没什么难题哈哈哈（主要是可以白嫖大赚特赚）。我自己研究研究这个问题吧，之前达尔优A98也出现-/_和=/+键chrome和打字上没问题，游戏死活没反应。最后给大家发了个新驱动解决了，我去打听打听~也可能是我自己硬件的问题，感谢您的解答和建议哈哈哈~

Gralerfics, 2022-08-25:
> 你提到的`remapBuffer[i] |= 0x80 >> j`确实是或，不干扰其他的，但后面还有一段hidBuffer相关的代码：
>
> ```
> index = (int16_t) (keyMap[_layer][i * 8 + j] / 8 + 1);
> bitIndex = (int16_t) (keyMap[_layer][i * 8 + j] % 8);
> ……
>
> if [...]

感谢解答，之前是忽略了后面的层 hhh，到手再试试了。祝好。

MuYunyun, 2022-10-08:
@JiumiYao @0xCC0x01 羡慕大佬们，个人硬件方面小白，硬件部分是否可以帮忙开发(有偿)或者推荐啥上手文章。。个人想先实践下软件方面部分。。

peng-zhihui, 2022-10-11:
已经修复了

closed 2022-10-11 by peng-zhihui