peng-zhihui/HelloWord-Keyboard #81: HID report发送之前对按键按下的判断？

0xCC0x01, 2022-08-19:
请教个问题：
发送HID report之前，没有判断是否有按键按下，这样即便任何按键都没按，也会固定发送，1ms一次
是否应该加上判断，有按键再发送report？

extern "C" void OnTimerCallback() // 1000Hz callback
{
    keyboard.ScanKeyStates();  // Around 40us use 4MHz SPI clk
    keyboard.ApplyDebounceFilter(100);
    keyboard.Remap(keyboard.FnPressed() ? 2 : 1);  // When Fn pressed use layer-2

    // Report HID key states
    USBD_CUSTOM_HID_SendReport(&hUsbDeviceFS,
                               keyboard.GetHidReportBuffer(1),
                               HWKeyboard::KEY_REPORT_SIZE);
}

fridyss, 2022-08-19:
> 

没按键按下时，需要清发0， 清除端口缓冲，不然主机会认为端口有数据， 会一直请求数据。

0xCC0x01, 2022-08-22:
> > 
> 
> 没按键按下时，需要清发0， 清除端口缓冲，不然主机会认为端口有数据， 会一直请求数据。

再请教下，是发一次0，清除缓冲就可以，还是在没有按键按下时，持续发0？

closed 2022-08-22 by 0xCC0x01

peng-zhihui, 2022-08-22:
HID是中断传输，主机定时读取数据的，有没有按键都需要上报，建议了解一下USB HID相关协议和报文格式。