peng-zhihui/HelloWord-Keyboard #93: 关于在翰文87的原理图基础上扩展配列的问题请教

zhdjet, 2022-08-23:
@peng-zhihui
您好,如果我想在此工程的基础上扩展配列;在瀚文键盘的原理图的基础上增加HC165的数量,请教增加的HC165应该要串连在哪个位置?

1.直接串连在XW06A的6个触摸按键的后面;

2.串连在XW06A的6个触摸按键前面(即键盘配列和触摸按键的中间);

3.直接在原理图的最前面增加.

谢谢!

peng-zhihui, 2022-08-23:
都可以，PCB的165芯片连接顺序和按键走线连接都可以是随机的，连完之后修改代码里的0层映射就行；不过从功能统一性角度来说，【2】是最合理的。

zhdjet, 2022-08-23:
谢谢!
另外再请教一下:在工程代码的0层映射中的第83个键(此层的最后一个[82]:85),实际键盘配列是82个键位,那这个85是不是6个触摸按键中的1个?工程中触摸按键有6个键位,在Smart Keys中有定义,但是工程中并未使用;这个6个触摸键位是不是可以分别定义使用,还是定义成一个整体使用?

zhdjet, 2022-08-23:
还有一个关于增加HC165扩展配列的问题:HC165悬空的引脚(HC165的IO总数大于实际配列键位数量)对键位的扫描有影响吗?谢谢!

xingrz, 2022-08-23:
> HC165悬空的引脚(HC165的IO总数大于实际配列键位数量)对键位的扫描有影响吗?

我之前用面包板随便试了一下，悬空不上拉的话读到的是低电平（相当于按下状态）。没仔细看代码，不确定不加0层映射的情况下一直处于低的位会怎么处理。但是既然用排阻的话，我觉得把不用的引脚都上拉也没坏处 😎

zhdjet, 2022-08-24:
谢谢!
假设我做了一个17键的数字小键盘(可拆装的),0层也映射了,如果我把数字小键盘拆了不装,只用主体键盘,那0层映射层是不是还要重新定义一下?如果不定义会不会有影响?

xingrz, 2022-08-24:
> 如果我把数字小键盘拆了不装,只用主体键盘,那0层映射层是不是还要重新定义一下?如果不定义会不会有影响?

从原理上分析：取决于主体上最后一个HC165的SI脚的电平（抱歉，现在手边暂时没资料，不确定悬空的情况下它是低还是高）。如果默认是高就还好，相当于没按下；如果默认是低那就相当于小键盘全被按下了。我觉得可以给最后一个SI脚加个上拉。

peng-zhihui, 2022-08-24:
不影响，所有按键都有排阻上拉，不接按键就是默认扫出来高电平（在代码中做了位反转所以会得到0）

zhdjet, 2022-08-24:
> 不影响，所有按键都有排阻上拉，不接按键就是默认扫出来高电平（在代码中做了位反转所以会得到0）

那如果我在此工程原理图的的6个触摸按键后面,再串联2片HC165来扩展小数字键盘,这个小键盘是分离式的,通过接口与键盘主体对接,而这2片HC165是焊在小键盘的PCB上的,工程代码中肯定是要把小数字键盘的键位在0映射区中映射的,如果小数字键盘不与键盘主体连接时,即只使用键盘主体,也就是主电路中不存在后扩的2片HC165,

1.此时0映射区要重新定义吗?

2.对整体的键位扫描会有影响吗?

3.如果上面这种情况对键位扫描有影响,请问在硬件电路或工程代码上要怎么处理?

谢谢!

peng-zhihui, 2022-08-24:
扫多少按键是代码决定的，主控并不知道你接了多少165，SPI给一个时钟脉冲就会读出一位信号，所以你把最后一个165的级联引脚拉高就行了，相当于不管主控读几个按键，只要超过实际物理按键数量，结果就都是1.

zhdjet, 2022-08-24:
好的,谢谢指教!我再琢磨琢磨;这两天把板子画好,有了实物再请教.非常感谢!

1074566721, 2022-09-20:
你好，关于更改配列的事，我们可以合作一下，我负责修改外壳模型，你改pcb，如果有意请添加我的qq 1074566721我们详聊

1074566721, 2022-09-20:
> 好的,谢谢指教!我再琢磨琢磨;这两天把板子画好,有了实物再请教.非常感谢!

你好，关于更改配列的事，我们可以合作一下，我负责修改外壳模型，你改pcb，如果有意请添加我的qq 1074566721我们详聊

STAR5847, 2023-02-24:
您好，我想在U10的第4脚第3脚第14脚第13脚第12脚第11脚上接出6个键位，这样就凑出了87个键位，请问可以怎么接嘛？