peng-zhihui/HelloWord-Keyboard #95: 主控板PCB直接在立创EDA里点PCB下单，打出来的板子轴体和rgb灯的孔都没开

Jack918, 2022-08-27:
[image]

嘉立创的回复是

[image]

是我下单的时候设置有问题吗

yangbo254, 2022-08-27:
专门去确定了下，目前立创EDA开源的地址，[稚晖](https://github.com/peng-zhihui)确实把led、轴孔的开孔画在了outline层，然后在outline2层又画了个只有外形的框，在机械层drl，仅画了你图片板子里面目前开孔的孔。
根据嘉立创19年出的免责规则，就是有drl认drl不认多outline的这条规则，看起来还真是稚晖画错了。。。

peng-zhihui, 2022-08-27:
这个是下单时间比较早吧？后面大量的单都是用这个文件做的，嘉立创应该是知道的。不过确实是第一次用立创EDA做项目，玩得还不是特别熟，后续我修改一下加个注释。

zhangsanbin, 2022-08-28:
建议花 3 元，勾选确认生产稿，再进行二次检查。

Chenshengshu, 2022-08-28:
我也遇到这个问题了，没做修改孔没开

QGB, 2023-08-22:
[image]
现在没有顺丰包邮了

QGB, 2023-08-22:
https://www.jlc.com/portal/q7i41203.html