peng-zhihui/HelloWord-Keyboard #101: 关于通信方面

371979859, 2022-09-01:
请问：
1、关于扩展版是如何设置电机的参数是直接单独调试好后保存在mcu里，还是可以使用usb通信来进行调整呢？后续会添加吗？
2、墨水屏的的图片数据上位机是怎么样传输的呢，目前usb口只有这一个功能吗？后面是否会开源上位机源码呢？
3、关于扩展版中使用的通信协议，个人感觉有点难理解，是否有什么学习的途径呢？
4、上位机只使用了libusb库吗？

还望大佬指点一二，谢谢！

peng-zhihui, 2022-09-02:
1.保存flash里面的，源码里有模拟eeprom的代码，只是还没有调用，可以自己看一下
2.usb bulk传输，会开放SDK代码
3.零基础的话去看《圈圈带你学USB》入门一下USB通信
4.目前是，后期会有很多三方库比如Qt、OpenCV之类的

371979859, 2022-09-02:
> 1.保存flash里面的，源码里有模拟eeprom的代码，只是还没有调用，可以自己看一下 2.usb bulk传输，会开放SDK代码 3.零基础的话去看《圈圈带你学USB》入门一下USB通信 4.目前是，后期会有很多三方库比如Qt、OpenCV之类的

谢谢稚晖君的回复

但是关于第三点其实我想问的是关于protocol.c/.hpp中使用的传输json格式数据的协议
我试着在PC端使用usb bulk发送“test_function”但是并没有反应，是否是我的使用方式错了呢？

另外关于打印浮点数的问题，在更改了CMakeList.txt后仍然无法使用printf直接打印，是否有另外的解决方法呢？

期待项目的后续发展！

DOCTOROLIVE, 2022-09-16:
> 

<圈圈教你玩USB>好像，哈哈

closed 2022-09-19 by 371979859