peng-zhihui/HelloWord-Keyboard #105: 关于扩展配列问题的请教

zhdjet, 2022-09-05:
@peng-zhihui
您好,现在扩展工程已经焊接调试完成,遇到如下几个问题请教:

1.第0号健位不响应(即扫描的起始键)

2.KeyMap的顺序只能在瀚文工程的基础上往下接,而且得照着上面Keycode定义好的键名增加,如果实物键盘的键位布局重新排列0层\1层\2层,实际键盘的输出跟实际键位编号是乱掉的.(感觉它是按照一个设定的标准布局走的)不能随意打乱)

3.关于Keycode数组的键名的定义,有没有参考的定义表或资料?假如我想在数字小键盘上增加:
一对小括号为一个键,
一对大括号为一个键,
一个打开计算器的快捷键,
一个分号键
或者再增加一些特殊的键值,要怎么处理?

4.增加FN键控制F1-F12定义多层控制功能,比如FN+F1打开浏览器等,要在哪里定义?

5.如果不要左侧的动态模组,可以增加旋转编码器控制音量或多媒体吗?

感谢百忙之中抽时间解惑答疑!

peng-zhihui, 2022-09-05:
1.早就已经修复了啊，pull最新代码
2.完全可以打乱，仔细理解一下代码
3.代码里有注释的，/**/里面的就是键名
4.加个if自己写个判断就行了
5.可以

zhdjet, 2022-09-06:
> 1.早就已经修复了啊，pull最新代码 2.完全可以打乱，仔细理解一下代码 3.代码里有注释的，/**/里面的就是键名 4.加个if自己写个判断就行了 5.可以

@peng-zhihui

大神,您好!

我现在把rgb灯取消了,增加了一个CapsLock的指示灯,就是在PB6引脚上接了下人LED,CapsLock按键的是按照FN键的方式接到PB1引脚上的,现在想切换大小写的时候这个指示灯能亮灭,指示键盘大小写的状态;
可是我憋了一天一夜也没写出来控制代码(照着HAL普通的按键控制灯的方式写的,但是不行),您能不能抽时间帮我写这个键盘按键控制MCU引脚上的LED的函数例程,我参考学习下下,非常感谢!!