peng-zhihui/HelloWord-Keyboard #118: 关于CapsLock键（Numlock键）功能实现的问题

zjzcpj, 2022-10-09:
问题描述：目前正在修改98配列的键盘，需要实现NumLock键的功能。个人感觉和CapsLock键功能的实现是类似的，但是目前还没理解固件代码中是如何实现大小写锁定功能的，希望得到一些指导，主要问题如下：
1、如何实现NumLock功能？（或者说原本是怎么实现CapsLock功能的？）
2、固件代码‘hw_keyboard.h’文件中有一行与大小写相关的变量声明“ bool isCapsLocked = false;”，但未查找到用法，请问这个变量的作用是什么？
3、目前打算将右下角的3个LED灯用作Caps、Num、Pause状态指示灯，请问该如何获取目前的”Caps、Num、Pause“状态？

zhdjet, 2022-10-11:
1.Num_lock功能,程序有的,增加数字键盘配列后,Num_lock解锁后,就是方向键和功能键;
2.关于指示灯的问题,灯和按键连接到MCU的GPIO上,用按键控制灯的方式控制,至于锁定状态,我也在研究中;

peng-zhihui, 2022-10-11:
1. 各种Lock功能不是键盘实现的，而是电脑实现的，键盘只负责触发对应按键。（你可以在电脑上插两把键盘，按下一把的CapLock之后，另一把打字也是大写锁定的）
2. 那个变量就是判断是否按下Lock，而这个信息是电脑下发过来的。
3. Lock信息是USB接收到的，在CUSTOM_HID_OutEvent_FS函数里收到的第一个uint8_t的参数就是电脑下发的lock信息，每一位代表一个Caps、Num、Pause的状态（具体哪个位是哪个你可以自己测试一下），判断完了控制对应的灯就行了
4. 发送对应的Lock按键就用代码里的Press函数就行，比如keyboard.Press(HWKeyboard::CAP_LOCK)