peng-zhihui/HelloWord-Keyboard #126: 6.6.1版本的cubeMX不再支持Clion开发

sytnocui, 2022-10-31:
最近手贱更新了一下cubeMx版本，发现已经无法打开之前 toolchain ide 是sw4stm32 的ioc文件了，也包括很多稚晖君项目的ioc文件，请问稚晖君这个有解决方案吗，还是说只能cubemx 降版本处理呢？
[image]

zhdjet, 2022-11-18:
clion是支持CubeMX6.6.1的，我现在就是用最新Clion和CubeMX的。

sytnocui, 2022-11-18:
> 
请问您使用cubemx生成的工程是sw4stm32的吗，我现在打开cubemx已经无法生成sw4stm32的工程了，而且之前创建的sw4stm32的工程也全都打不开了

zhdjet, 2022-11-18:
stm32cubeIDE

zhdjet, 2022-11-18:
.ioc格式应该是CubeMX全版本兼容的；生成的代码工程文件格式是给编译IDE用的。

sytnocui, 2022-11-19:
> .ioc格式应该是CubeMX全版本兼容的；生成的代码工程文件格式是给编译IDE用的。

确实，我试了一下新建的cubemx工程如果选择toolchain是stm32cubeide的话，clion打开的，编译，单步调试确实都没有问题。
看来这样确实可以继续使用clion开发

但是有一个巨大的问题，您有没有试过使用6.6.1版本的cubemx打开过稚晖君之前的很多项目的ioc文件，稚晖君之前很多项目工具链选择都是sw4stm32，我尝试使用6.6.1版本的cubemx打开之前版本创建的工具链选择为sw4stm32的，都会报错打不开，就是最上面那个报错图，这意味着稚晖君很多项目的ioc文件我都看不了了，所以我现在又退回了6.5.0版本的cubemx。

今后我可能创建工程toolchain都会选择stm32cubeide，然后继续使用clion开发了，不知道稚晖君对这个问题怎么看？
@peng-zhihui

zhdjet, 2022-11-21:
其实是有一个非常简单的方法：
就是把旧版本的工程.ioc文件中的关于旧版本中关于版本号的数据全部改成新版本的；
再把sw4stm32改成stm32cubeide，保存后就可以用新版本的CubeMX打开了。
这个方法我一般不知诉别人的，哈哈。。。

zhdjet, 2022-11-21:
另外其实关于CubeMX的版本号不改，把sw4stm32改成stm32cubeide，也能打开；但是会有一个选择版本的提示框，随便点一个，就能打开。