peng-zhihui/HelloWord-Keyboard #135: 关于瀚文键盘休眠后全键唤醒方案讨论以及HID键盘报文17字节切换标准8字节的请教

Wangzm007, 2022-11-28:
1.瀚文原连接方式想要唤醒休眠状态的话需要像Fn键一样,一个I/O连接一个按键上拉端,但是I/O口有限无法全键唤醒.
[image]
1.1所以我想到如下图两种连接方式来解决:方案①将原来按键的接地端统一接到单片机的一个I/O口上,该I/O口常态输出低电平,在进入休眠状态时重新初始化为输入状态,使能中断;在中断处理程序中退出休眠状态以及重新初始化为输出低电平状态.
1.2方案②由①变化而来,A口专门控制CMOS的连通与断开,B口中断输入判断:正常状态下用A控制CMOS连通,B口关闭中断使能;进入休眠时断开CMOS以及使能B中断输入使能.退出休眠时切换状态即可.
[image]
1.3上诉两种方案的不足之处在于休眠后第一次唤醒的按键可能不会输出到PC端(上述方案仅做讨论,未验证)
1.4写到这儿临时有个不成熟的想法,就是瀚文原方案中74HC165在单片机休眠时处于什么状态?按下某个按键时,74HC165是否会输出脉冲或电平转换到MCU,是的话能否将MCU相关的I/O口改为中断输入,正常状态初始化为SPI,休眠状态下初始化为中断输入以实现全键唤醒(瞎想的,我的编程水平无法验证)
-===================================================================
2.了解到一般HID键盘为了兼容在BIOS界面下使用是8字节报文(6键无冲),即第3字节:0x04是a,0x05是b,0x06是c ...等等;瀚文键盘17个字节按位表示(全键无冲),第三字节:0x10是a,0x20是b,0x40是c ... 等等.  但是看见有评论说瀚文键盘在bios界面下是可以使用的,所以想请教UP一下,瀚文是否可以切换这两种格式,让瀚文在有线连接进入系统时切换到1ms硬报文模式,其他状态下（如BIOS、无线蓝牙、2.4g连接时）切换到8字节报文兼容模式；如果可以的话应该在哪里切换,或者我想增加8字节报文的方式应该在程序哪个地方修改呢？
    
    期待up或者其他大佬有空的时候能够回复解答一下哦~主要是第2点报文问题，程序上我确实不太懂，买了最新版的00教你玩usb啃的也很艰难,万分感谢~
[image]

CC52242, 2022-11-30:
关于第一点的键盘唤醒，74hc165需要mcu为其提供时钟信号才能输出数据，所以mcu在休眠状态下无法通过读165来唤醒。
你可以看下我提的这个issue[https://github.com/peng-zhihui/HelloWord-Keyboard/issues/132](url).

关于第二点的usb报文，需要区分的是主机状态而不是连接方式。如果主机在UEFI或系统内，那就1kHZ；如果在Bios,那就8字节兼容。这大概只适用于采用usb连接时。2.4G以及BLE连接需要考虑的就不止usb报文的问题了。
对主机状态的区分可能需要通过区别主机发的控制传输来实现？
实在不行那就上电第一次配置usb设备时把自己设成8字节，进入系统后软重启重配置为1kHZ

我自己也是正在学习，意见中可能会有不少荒唐的地方，还请赐教指正

xingrz, 2022-12-19:
我觉得楼主这个想法很合理，应该是可以实现的。甚至更进一步，我觉得可以加一个 MOS 管休眠状态下把 74HC165 的 VCC 全断掉（不过这个得测一下 74HC165 的电流多大，如果确实小得忽略不计，那就没必要了）。

> 就是瀚文原方案中74HC165在单片机休眠时处于什么状态?按下某个按键时,74HC165是否会输出脉冲或电平转换到MCU

165 是不会主动输出电平的，得靠 MCU 的时钟信号驱动它。

BrianLChen, 2023-01-09:
165再无时钟信号的情况下不会输出，但是应该是有输入的吧。如果从休眠状态回到正常状态的时间足够短，是不是165依然可以检测到按键？也就是说可能并不会吞掉唤醒按键。
第二种方法做唤醒应该是没有问题的，B口做为输入，自身下拉，高电平唤醒。

Mexico-zyy-lost, 2023-01-11:
usb的报文协议在哪可以看？

ourcabin0746, 2023-01-31:
我也一直在思考休眠的问题，请问大家有可用的思路和方法了吗？
在USB可以考虑不用休眠，但是如果瀚文这套方案用在BLE，2.4G下，麻烦就比较多。

Xing-C, 2024-03-24:
有哪位大佬有进展了吗？  迟迟未动手的原因之一就是想有个2.4G...

BrianLChen, 2024-03-26:
> 有哪位大佬有进展了吗？ 迟迟未动手的原因之一就是想有个2.4G...

用nrf52840写了一个，可以在我仓库里看到。但是有一些bug，一直没时间修

lijunru-hub, 2024-06-26:
用 esp32s3 做了一个键盘 https://oshwhub.com/esp-college/esp-keyboard，全按键唤醒，低功耗 BLE等

TheShallow, 2024-06-26:
这是来自小项的自动回复邮件。
 
邮件已收到

lijunru-hub, 2024-06-26:
USB 的全键无冲和六键的上报逻辑是。

如果你按下的按键没超过 6 个，就按六键的接口上报，超过就按全键的接口上报。
在 BIOS 里面因为不会去识别 USB HID 报文，所以键盘里面默认有一个标准六键上报的就行了，只要你不按太多个键，BIOS 里面就是能正常用的