peng-zhihui/HelloWord-Keyboard #154: 更新墨水屏图片失败！驱动已安装完成

CrazyzhouLL, 2023-03-17:
使用software工具更新墨水屏图片失败！驱动已安装完成，
求助怎么解决。。。
报错截图
[image]
设备管理界面截图
[image]

remador, 2023-04-12:
Estou a 4,5 meses nessa tortura.
Nem tenho palavras pra agradecer,
Muito obrigado!
@Mitodacaverna

TheShallow, 2023-04-12:
这是来自小项的自动回复邮件。
 
邮件已收到

remador, 2023-04-12:
Muito obrigado!
Estou exausto!
@remador

remador, 2023-04-12:
Muito obrigado!
Estou exausto.
@remador

XBMAX, 2024-04-14:
I just built my keyboard and I'm a bit confused how to get the e-ink screen working? My screen isn't working out of the box. I have downloaded the actual file and the first test image but it says failed to connect to the device. I do have Zadig and the STM program - can someone please assist me in what I need to do to get this function? Also is there a .json file to use with VIA? 

Thank you!