peng-zhihui/HelloWord-Keyboard #156: MUTE，VOLUME_UP，VOLUME_DOWN这类功能键不起作用

joye-wang, 2023-04-02:
请教一下，为什么 MUTE，VOLUME_UP，VOLUME_DOWN这类按键不起作用呢？
普通按键都是可以映射的，类似A-Z,  F1-F12都是可以的。但是从键盘枚举的某一个位置开始之后所有的按键都没用了.
请看代码的注释：
```
enum KeyCode_t {
	/*------------------------- HID report data -------------------------*/
	LEFT_CTRL = -8,
	LEFT_SHIFT = -7,
	LEFT_ALT = -6,
	LEFT_GUI = -5,
	RIGHT_CTRL = -4,
	RIGHT_SHIFT = -3,
	RIGHT_ALT = -2,
	RIGHT_GUI = -1,

	RESERVED = 0,
	ERROR_ROLL_OVER,
	POST_FAIL,
	ERROR_UNDEFINED,
	A,
	B,
	C,
	D,
	E,
	F,
	G,
	H,
	I,
	J,
	K,
	L,
	M,
	N,
	O,
	P,
	Q,
	R,
	S,
	T,
	U,
	V,
	W,
	X,
	Y,
	Z,
	NUM_1/*1!*/,
	NUM_2/*2@*/,
	NUM_3/*3#*/,
	NUM_4/*4$*/,
	NUM_5/*5%*/,
	NUM_6/*6^*/,
	NUM_7/*7&*/,
	NUM_8/*8**/,
	NUM_9/*9(*/,
	NUM_0/*0)*/,
	ENTER,
	ESC,
	BACKSPACE,
	TAB,
	SPACE,
	MINUS/*-_*/,
	EQUAL/*=+*/,
	LEFT_U_BRACE/*[{*/,
	RIGHT_U_BRACE/*]}*/,
	BACKSLASH/*\|*/,
	NONE_US/**/,
	SEMI_COLON/*;:*/,
	QUOTE/*'"*/,
	GRAVE_ACCENT/*`~*/,
	COMMA/*,<*/,
	PERIOD/*.>*/,
	SLASH/*/?*/,
	CAP_LOCK,
	F1,
	F2,
	F3,
	F4,
	F5,
	F6,
	F7,
	F8,
	F9,
	F10,
	F11,
	F12,
	PRINT,
	SCROLL_LOCK,
	PAUSE,
	INSERT,
	HOME,
	PAGE_UP,
	DELETE,
	END,
	PAGE_DOWN,
	RIGHT_ARROW,
	LEFT_ARROW,
	DOWN_ARROW,
	UP_ARROW,
	PAD_NUM_LOCK,
	PAD_SLASH,
	PAD_ASTERISK,
	PAD_MINUS,
	PAD_PLUS,
	PAD_ENTER,
	PAD_NUM_1,
	PAD_NUM_2,
	PAD_NUM_3,
	PAD_NUM_4,
	PAD_NUM_5,
	PAD_NUM_6,
	PAD_NUM_7,
	PAD_NUM_8,
	PAD_NUM_9,
	PAD_NUM_0,
	PAD_PERIOD,
	NONUS_BACKSLASH, // 大概从这个位置开始，下面的按键都没用
	APPLICATION, 
	POWER,
	PAD_EQUAL,
	F13,
	F14,
	F15,
	F16,
	F17,
	F18,
	F19,
	F20,
	F21,
	F22,
	F23,
	F24,
	EXECUTE,
	HELP,
	MENU,
	SELECT,
	STOP,
	AGAIN,
	UNDO,
	CUT,
	COPY,
	PASTE,
	FIND,
	MUTE,
	VOLUME_UP,
	VOLUME_DOWN,
	FN = 1000
/*------------------------- HID report data -------------------------*/
};

```

yr785444662, 2023-04-17:
同上，媒体功能按键无效

BrianLChen, 2023-05-14:
Windows 好像 HID Usage ID 好像到101就没有了，想要实现音量控制需要使用 Consumer Page （0x0c）中的 Volume Control 才可以实现。我也在尝试中。

BrianLChen, 2023-06-18:
可以在原先的report desc中加入一个新的collection， Report ID = 2；
report buffer 的第1个字节用于声明report ID（我记得我在代码里看到有这部分的了）。

        /* Volume */ \
        0x05, 0x0c,  /* Usage Page (Consumer) */ \
        0x09, 0x01,  /* USAGE Consumer Control */ \
        0xa1, 0x01,  /* COLLECTION (Application)                       */ \
        0x85, 0x02,  /* Report ID = 2 */ \
        0x05, 0x0c,  /* Usage Page (Consumer) */ \
        0x15, 0x00,  /* logic min */ \
        0x25, 0x01,  /* logic max */ \
        0x75, 0x01,  /* rep size 1 */ \
        0x95, 0x08,  /* rep count (07) */ \
        0x09, 0xe9,  /* usage volume increase */ \
        0x09, 0xea,  /* usage volume decrease */ \
        0x09, 0xe2,  /* usage mute */ \
        0x09, 0xcd,  /* uasge play/pause */ \
        0x09, 0xb5,  /* uasge next track */ \
        0x09, 0xb6,  /* uasge Previous track */ \
        0x09, 0xb7,  /* uasge stop */ \
        0x0a, 0x92, 0x01, /* Open Application: Calculator */ \
        0x81, 0x02,  /* output */ \
        \
        0xc0,                           /* END_COLLECTION                                 */\

需要对原本代码中的remap的结构进行修改，对 report ID = 1 和 report ID = 2 的两个 report buff 进行分开计算。优先对 report ID = 1 的report buffer进行报表（考虑到键盘时更加重要的功能）。当 report ID=1 所对应的 report buffer 没有改动时，对report ID = 2 的 report buffer 进行报表，所以需要两个不同的 report buffer。

所以需要额外的判断，判断2个report buffer 是否较之前有所变化。

两个不同report ID只能进行分开的报表。而键盘的usage type时keyboard，音量控制的usage type 时consumer control， 不同的usage type也不能放在同一个collection中。

好像还有一种方法就是重新声明一个USB HID 设备，使用不同的endpoint。目前并没有对这个方法进行深入的研究