peng-zhihui/HelloWord-Keyboard #163: Type-C电源供电和发热问题

ideal1213123, 2023-07-06:
单独 TYPE-C 供电，主控板 RGB 灯珠后半部分会闪烁。加上电池同时供电后 RGB 灯珠正常发光，但 Type-C PCB 板严重发热，无扩展模块。请问这正常吗

TheShallow, 2023-07-06:
这是来自小项的自动回复邮件。
 
邮件已收到

PelloBravo, 2023-07-07:
这个方案应该是不能用typec直连的，应该用usb2.0的typeA接口转typec。
我猜测是因为加入了上拉电阻的缘故，导致usb3的typec输出12v电压导致的电路板发热。

ideal1213123, 2023-07-16:
> 

所以加上电池供电后发热是可以用的，不会烧掉。对吧。

PelloBravo, 2023-07-16:
难说，理论上是可以，因为原设计MCU是信号线供电的。不过你typec和电池同时连是怎么连的？串联还是并联？

xingrz, 2023-07-31:
> 这个方案应该是不能用typec直连的，应该用usb2.0的typeA接口转typec。 我猜测是因为加入了上拉电阻的缘故，导致usb3的typec输出12v电压导致的电路板发热。

[image]

我特意去重新看了眼原理图，USB-C 口只有 CC1 和 CC2 分别 5.1K 下拉，没有你所说的上拉。这是符合 USB-C 的规范的。

另外，无论 USB 2.0 还是 3.0，无论 Type-A 还是 Type-C，只要你没有专门的芯片去做 USB-PD 握手，VBUS 上只会是 5V。并不会像你所说加了电阻就变成了 3.0、3.0 就是 12V 的情况。

PelloBravo, 2023-08-01:
啊对，是我接线错了。。。我加了一个ph2.0的端口接买来的typec口调试，那个typec口是din dout+cc线的不是+uart的。。。
不过我接usb3.0后测出的LED电压确实是12v，可能是我的线自带PD芯片？

xingrz, 2023-08-01:
> 我加了一个ph2.0的端口接买来的typec口调试，那个typec口是din dout+cc线的不是+uart的

没看懂…感觉你把不同的东西混为一谈了。USB 2.0 就 VBUS/D-/D+/GND 四根线，哪怕是 USB 3.0 多了四对 TX/RX 也是 USB 信号，并不是 UART，也不叫 DIN DOUT。

我猜，你说的 `买来的typec口` 并不是单纯的 USB Type-C 连接器，而是别人做好的某种转接板？

PelloBravo, 2023-08-01:
不是，是6pin的ph2.0转typec，6pin其中是d+d- cc1cc2 gnd vbus，要想正常使用得是d+d- tx rx gnd vbus。
typec的d+d-就是2.0的信号

PelloBravo, 2023-08-01:
我看了下原理图，uart是控制那个dynamic模块的，实际上typec接口只接了4根线。。