peng-zhihui/HelloWord-Keyboard #168: Pcb files again

KO-07, 2023-09-12:
Hi, I just wanted to know if there are any pcb files to this project, or is there a way to convert the pdf schematics into the herber file format?

Thanks.

TheShallow, 2023-09-12:
这是来自小项的自动回复邮件。
 
邮件已收到

KO-07, 2023-09-12:
I don't speak Chinese my friend :(

KO-07, 2023-09-12:
Is there a link to where the PCBs are for this project?

realhistory, 2023-09-21:
> Is there a link to where the PCBs are for this project?
is here : https://oshwhub.com/pengzhihui/b11afae464c54a3e8d0f77e1f92dc7b7

realhistory, 2023-09-21:
> The website is in Chinese, and its asking me to create an account in order to download it.
> 
> Is there anyway someone who has already Dowland it could attach it?
> 
> Sorry for any inconvenience that this may cause.

sure,please leave your email, i will send to you files

KO-07, 2023-09-21:
Sure thing, I will attach my email, but before that, if it's ok, I just wanted to know if there is any other way to attach the files to github?

KO-07, 2023-09-21:
Anyway to send besides email if possible? If not then I'll give you the email address.

maghdzany, 2024-03-23:
> > The website is in Chinese, and its asking me to create an account in order to download it.
> > Is there anyway someone who has already Dowland it could attach it?
> > Sorry for any inconvenience that this may cause.
> 
> sure,please leave your email, i will send to you files

Can you also send me gerber files? This is my email maghzml@gmail.com I can't login with wechat :')

KO-07, 2024-03-23:
Hi, I would also like the gerber files/pcb files for this project. If anyone could send me the files to my email, that would be much appreciated.
KO7.Fzn@Gmail.com

cretator, 2026-11-07:
It’s been a while; I don’t know if anyone will see this. fergringasho12@gmail.com